atsams-rs/atsamx7x-rust #14: HAL: Upstream from OVL works

jamesmunns, 2022-04-22:
Hey, I've been working on ATSAME70 support specifically in a repo as part of some contracting work (OVL => OneVariable UG Labs).

Since I've been focused on just this one chip, and a relatively limited set of features, it's probably not ready to merge 1:1, but just in case someone is interested, the [repo is here](https://github.com/jamesmunns/same70-experiments/tree/main/vendor/atsamx7x-hal), and I've published the crate under the name `ovl-atsamx7x-hal` for now, until this work gets upstreamed.

So far, I have the following components implemented:

* EFC - basically just the ability to set wait states
* GMAC - some corners rounded, really only tested to work with the KSZ8061 PHY, but capable of "doing ethernet" and has a `smoltcp` implementation capable of maintaining a TCP connection. Only supports one set of pins ATM
* PIOx - generally all 5 ports are supported, and some pin functionality is working (mostly setting ABCD peripheral modes, and output pins)
* PMC - Pretty rudimentary support, but enough to set some prescalers. I mostly use it to reach a single clock config (300MHz CPU clock, 150MHz MCK)
* Serial - I haven't really touched the upstream code here, I'm not using serial
* SPI - Basic "transfer" API support for the SPI0 peripheral. Only supports one set of pins ATM
* WDT - Basically just enough to shut it off

Feel free to take any of this work if you want/need it before I get a chance to upstream! I've kept the 0BSD license, so you can do what you'd like.

jamesmunns, 2022-04-22:
Oh, there are also quite a few working examples in the `same70-bringup` project, [available here](https://github.com/jamesmunns/same70-experiments/tree/main/firmware/same70-bringup/src/bin).

tmplt, 2022-04-22:
Thanks for the notice. I'll take a more thourough look at it when able.

michalfita, 2022-09-16:
@tmplt is there anything from that @jamesmunns work here that wasn't in #25?

tmplt, 2022-09-16:
Only the GMAC and `groundhog::RollingTimer` implementations. I'll push a WIP PR for the former tomorrow. I do not consider the latter in-scope for the HAL currently.

tmplt, 2022-09-17:
Superseded by #33.

michalfita, 2022-09-21:
The #33 is a PR, as well as #32 and #31 which are continuation of work from #24.