atsams-rs/atsamx7x-rust #23: Move heavylifting to the user, relief us from updating over 40k files

michalfita, 2022-07-26:
As suggested by @tmplt in #22, this is about harnessing power of `build.rs` to manage PAC crates.

michalfita, 2022-07-26:
@tmplt I have good news: I've hacked my way to get PAC crate where `build.rs` handles generation via `svd2rust`. Some long standing unresolved problems in Rust itself stand in opposition to do it in a clean way, but in overall this works with some tradeoffs:
* All attributes have to be in the _static_ `src/lib.rs`, so cannot be generated by the tool; there's flag controlling this, so not a big deal apart keeping eye if that changes in the future.
* Regardless of the flag `svd2rust` leaves `#![doc = ...]` attribute in the generated `${OUT}/lib.rs` what causes error on inclusion; the workaround is to filter that out from the output, what's memory inefficient.

I'll work on this more and PR some draft, but for now I need to move back to issues that require closing ASAP.

michalfita, 2023-02-20:
It took some time, but most things seem to be in place. I've published [some alpha releases](https://crates.io/crates/atsame70n19b/0.28.0-alpha0), if anyone is willing to test using them from crates.io. FYI @tmplt 

BTW. I've noticed `atsam-rs` now publish PACs for SAME70 with `-pac` suffix. I wonder if that competition makes any sense.

tmplt, 2023-02-24:
On 02/20/2023 22:33, Michał Fita wrote:
> It took some time, but most things seem to be in place. I've published 
> some alpha releases 
> <https://crates.io/crates/atsame70n19b/0.28.0-alpha0>, if anyone is 
> willing to test using them from crates.io. FYI @tmplt 
> <https://github.com/tmplt>

I'll take a look once a PR is ready.

> BTW. I've noticed |atsam-rs| now publish PACs for SAME70 with |-pac| 
> suffix. I wonder if that competition makes any sense.

Looks like a side-effect of other PAC generation.

michalfita, 2023-02-28:
> Looks like a side-effect of other PAC generation.

They [responded to my request](https://github.com/atsam-rs/atsam-pac/issues/25#issuecomment-1437644032).

> I'll take a look once a PR is ready.

That's going to be separate repository (actually is, but not publicized yet).

michalfita, 2023-09-12:
This is addressed in [atsamx7x-pac](https://github.com/atsams-rs/atsamx7x-pac) repository and issue #56 / PR #61.

closed 2023-09-12 by michalfita