atsams-rs/atsamx7x-rust #26: Feature check in `hal/build.rs` fatally interferes with the `rust-analyzer`

michalfita, 2022-09-14:
**Caused by #25**

The check for missing feature in `hal/build.rs` is an impediment to operations of the `rust-analyzer`. I don't recall observing this problem when the check was in the code.

```
[ERROR rust_analyzer::lsp_utils] rust-analyzer failed to load workspace: Failed to read Cargo metadata from Cargo.toml file /home/manveru/Projects/atsamx7x-rust/Cargo.toml, Some(Version { major: 1, minor: 63, patch: 0 }): Failed to run `"cargo" "metadata" "--format-version" "1" "--manifest-path" "/home/manveru/Projects/atsamx7x-rust/Cargo.toml" "--filter-platform" "thumbv7em-none-eabihf"`: `cargo metadata` exited with an error: error: failed to load manifest for workspace member `/home/manveru/Projects/atsamx7x-rust/pac/template`

Caused by:
  failed to read `/home/manveru/Projects/atsamx7x-rust/pac/template/Cargo.toml`

Caused by:
  No such file or directory (os error 2)

[ERROR rust_analyzer::lsp_utils] failed to run build scripts

error: failed to run custom build command for `atsamx7x-hal v0.3.0 (/home/manveru/Projects/atsamx7x-rust/hal)`

Caused by:
  process didn't exit successfully: `/home/manveru/Projects/atsamx7x-rust/target/debug/build/atsamx7x-hal-c2579600962003a0/build-script-build` (exit status: 1)
  --- stderr
  Error: "The HAL is built for a specific target device selected using a feature, but no such a feature was selected."
```

tmplt, 2022-09-14:
Would adding a `rust-analyzer` configuration that builds the crate with the correct feature-set close this issue? Or is this not supported? The analyzer seems to be designed for editor invocations.

tmplt, 2022-09-14:
The `compile_error!` approach was replaced with `build.rs` back when some complexity was present to the feature flags. This has now been removed, so we can revert back to the former approach.

michalfita, 2022-09-15:
Makes sense. We'll figure something out - that has lower priority ATM than polishing the code from #25.

closed 2022-09-16 by michalfita