atsams-rs/atsamx7x-rust #34: Publishing the 0.3 versions of HAL and board crates

jelacious, 2022-09-28:
Many of the crates in this megaproject has seen a lot of changes since last publication on crates.io. I just assume this has slipped the mind of the maintainers. Latest version (0.1.0) was published in March. It'd be nice to have the 0.3 versions of the HAL and board crates be published, if they are ready ofc. All the peripheral access APIs have a version of 0.25.0 in their Cargo.toml files but the latest version published is 0.21.0 also back in March.

Looking forward to getting acquainted with this. Keep it up <3

michalfita, 2022-09-28:
It's work in progress, but I have limited capacity to touch this ATM. It should happen before end of this year.

PAC versioning is weird at the moment. We base it on minor version of `svd2rust`, so fluctuates as we're getting new releases.

jelacious, 2022-09-28:
Ah, that makes sense. 

No rush, will just pin myself against the repo itself for now.

tmplt, 2022-10-01:
> It'd be nice to have the 0.3 versions of the HAL and board crates be published

We don't offer proper BSPs yet, but I don't think it requires a lot of work to get there.

michalfita, 2022-10-07:
@jelacious I let myself to use this issue to address the release.

- [ ] Lying foundations
- [ ] Tagging `pac/v0.25.0`
- [ ] Publishing PAC crates
- [ ] Tagging `hal/v0.4.0`
- [ ] Publishing HAL crates

michalfita, 2022-10-07:
PAC crates v0.25.0 released to [crates.io](https://crates.io/keywords/atsam).

tmplt, 2022-10-12:
@michalfita, I suggest adding me as co-owner of repo and crates so there are multiple people that can handle releases.

michalfita, 2022-10-13:
Not finished.

commit 32935d57 (2022-10-12): [Fixes #34] Lying foundations for release
commit 0fa381e2 (2022-10-17): [Fixes #34] After PAC 0.25.0 some leftovers required addressing.

This would form the next release, so we can go with next releases with a clean slate.
closed 2022-10-19 by michalfita