atsams-rs/atsamx7x-rust #37: Configuring System IO Pins Fails Silently When Feature Is Not Used

martinmortsell, 2022-10-07:
The system IO pins, e.g. PB04 need to be handled by the MATRIX peripheral as well as the PIO peripheral.
This was feature gated behind the `configurable-system-pins` feature, as using these pins for other things could potentially brick your device.

However, attempting to configure said pins without the feature fails silently, where it should fail to compile or at the very least panic during runtime.

michalfita, 2022-10-12:
If I understand problem correctly it would need to be addressed by `svd2rust` and generate access as a function which for disabled feature would emit informative [`compile_error!()`](https://doc.rust-lang.org/std/macro.compile_error.html).

If that's the case we need to think about building up the case to request this.

michalfita, 2022-10-12:
OK, these looks like related to pins kept in HALs pin bank, but the method for more friendly errors would apply.

tmplt, 2022-10-12:
Solving this at the svd2rust level seems like more work than worth at the present. Best to postpone a more detailed compile error when there is an obvious approach. Proper documentation would cover it.

michalfita, 2022-10-24:
#38 fixes this.

closed 2022-10-24 by michalfita