atsams-rs/atsamx7x-rust #40: CI is Broken

martinmortsell, 2022-10-13:
The CI is broken since the last push.
The CI for #38 fails in a few places (where it decides to run).
First cause of failure is that the CI think every subfolder of `./pacs/` is a pac, it clears the two first characters `at` normally, and builds with the selected pac as a feature.
Now there is a `templates` folder resulting in an attempt to run `cargo build --features unproven,mplates`.
Second Cause of failure mostly speaks for itself:
```
`cargo metadata` exited with an error: error: current package believes it's in a workspace when it's not:
current:   /home/runner/work/atsamx7x-rust/atsamx7x-rust/boards/atsame70_xpro/Cargo.toml
workspace: /home/runner/work/atsamx7x-rust/atsamx7x-rust/Cargo.toml

this may be fixable by adding `boards/atsame70_xpro` to the `workspace.members` array of the manifest located at: /home/runner/work/atsamx7x-rust/atsamx7x-rust/Cargo.toml
Alternatively, to keep it out of the workspace, add the package to the `workspace.exclude` array, or add an empty `[workspace]` table to the package's manifest.
cargo-fmt 
```
and results from the changes made to `Cargo.toml` in the root folder.

martinmortsell, 2022-10-13:
Related  to this, when trying to figure out why #38 wasn't passing I found that a couple of the `atsame70_xpro` examples aren't building successfully.

tmplt, 2022-10-13:
32935d57f0a76bdddfb4243647de530d426436f9 broke CI. There was also no PR for it, from the looks of things.

michalfita, 2022-10-13:
I was making attempt to release PACs, but I couldn't. There were missing pieces of the puzzle, no Rust code changes. So I did my best to make this work bringing back some of my original ideas to the update script (originally in Python, rewritten to bash by @tmplt, but incomplete). I've managed to release PACs, but there were other issues, so I pushed what worked to develop directly so it sits there tagged.

Now, I have more changes to fix builds of other stuff and address CI, but I plan to PR these, so hopefully I'll manage to do this over weekend.

Then I'm afraid we have to scrap this repo in favour of separate repos for PACs and HAL, major reason is the size of the current one and problem with checkouts, the next reason is it would make it easier to release and tag releases per repository. I'll do this for PACs first with my new method of building them. Later I can extract full history for `hal` and `boards` directories only into the new repository. Both would free us from legacy of LFS and 40k PAC files.

tmplt, 2022-10-15:
> I was making attempt to release PACs, but I couldn't. There were missing pieces of the puzzle [...] [rewritten to bash but incomplete]

How so? What was missing?

> I'll do this for PACs first with my new method of building them.

What is this new method?

michalfita, 2022-10-17:
> How so? What was missing?

PACs have to have right `license = ` field in `Cargo.toml` and `README.md` has to be present in each. That's now fixed.

> What is this new method?

The crate doesn't deliver several thousands of files, but runs `svd2rust` and `form` as libraries in `build.rs` - I've tested this approach with `crates.io` and this seem to work pretty well.

tmplt, 2022-10-17:
> runs svd2rust and form as libraries in build.rs - I've tested this approach with crates.io and this seem to work pretty well.

Good work. Looking forward to see it merged.

commit 3387196c (2022-10-16): [Fixes #40] Amend `CHANGELOG.md`
commit 9c165cd3 (2022-10-17): [Fixes #40] Cleaning-up
closed 2022-10-19 by michalfita