atsams-rs/atsamx7x-rust #54: atsame70_xpro example fails to build with blinky_delay and blinky_irq

cquick01, 2023-05-05:
Getting an error when trying to build the `atsame70_xpro` example. The other `atsamv71_xult` one builds fine.

```
error: linking with `rust-lld` failed: exit status: 1
  |
  = note: "rust-lld" "-flavor" "gnu" "/tmp/rustcvue7Jj/symbols.o" "/home/cquick/atsamx7x-rust/boards/atsame70_xpro/target/thumbv7em-none-eabihf/debug/examples/blinky_delay-873d777b61c44224.atsamx7x_hal-d5a980c4e9a39481.101fkh0e4q7472nd.rcgu.o.rcgu.o" "--as-needed" "-L" "/home/cquick/atsamx7x-rust/boards/atsame70_xpro/target/thumbv7em-none-eabihf/debug/deps" "-L" "/home/cquick/atsamx7x-rust/boards/atsame70_xpro/target/debug/deps" "-L" "/home/cquick/atsamx7x-rust/boards/atsame70_xpro/target/thumbv7em-none-eabihf/debug/build/atsame70q21b-99c3868bec83c264/out" "-L" "/home/cquick/atsamx7x-rust/boards/atsame70_xpro/target/thumbv7em-none-eabihf/debug/build/cortex-m-bb856671e465605e/out" "-L" "/home/cquick/atsamx7x-rust/boards/atsame70_xpro/target/thumbv7em-none-eabihf/debug/build/cortex-m-rt-6a7ae1b79b44e4b7/out" "-L" "/home/cquick/.rustup/toolchains/1.63.0-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib" "-Bstatic" "/tmp/rustcvue7Jj/libcortex_m_rt-a8a12616f111b1f5.rlib" "--start-group" "/tmp/rustcvue7Jj/libcortex_m-aaa8d6483869b2d6.rlib" "--end-group" "/home/cquick/.rustup/toolchains/1.63.0-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib/libcompiler_builtins-814a2ee6da8588fb.rlib" "-Bdynamic" "--eh-frame-hdr" "-znoexecstack" "-L" "/home/cquick/.rustup/toolchains/1.63.0-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib" "-o" "/home/cquick/atsamx7x-rust/boards/atsame70_xpro/target/thumbv7em-none-eabihf/debug/examples/blinky_delay-873d777b61c44224" "--gc-sections" "-Tlink.x"
  = note: rust-lld: error: undefined symbol: _SEGGER_RTT
          >>> referenced by lib.rs:175 (/home/cquick/.cargo/registry/src/github.com-1ecc6299db9ec823/rtt-target-0.3.1/src/lib.rs:175)
          >>>               /home/cquick/atsamx7x-rust/boards/atsame70_xpro/target/thumbv7em-none-eabihf/debug/examples/blinky_delay-873d777b61c44224.atsamx7x_hal-d5a980c4e9a39481.101fkh0e4q7472nd.rcgu.o.rcgu.o:(rtt_target::UpChannel::conjure::hb1d03a52c6959c14)
          >>> referenced by lib.rs:175 (/home/cquick/.cargo/registry/src/github.com-1ecc6299db9ec823/rtt-target-0.3.1/src/lib.rs:175)
          >>>               /home/cquick/atsamx7x-rust/boards/atsame70_xpro/target/thumbv7em-none-eabihf/debug/examples/blinky_delay-873d777b61c44224.atsamx7x_hal-d5a980c4e9a39481.101fkh0e4q7472nd.rcgu.o.rcgu.o:(rtt_target::UpChannel::conjure::hb1d03a52c6959c14)
          >>> referenced by lib.rs:181 (/home/cquick/.cargo/registry/src/github.com-1ecc6299db9ec823/rtt-target-0.3.1/src/lib.rs:181)
          >>>               /home/cquick/atsamx7x-rust/boards/atsame70_xpro/target/thumbv7em-none-eabihf/debug/examples/blinky_delay-873d777b61c44224.atsamx7x_hal-d5a980c4e9a39481.101fkh0e4q7472nd.rcgu.o.rcgu.o:(rtt_target::UpChannel::conjure::hb1d03a52c6959c14)
          >>> referenced 5 more times

error: could not compile `atsame70_xpro` due to previous error
warning: build failed, waiting for other jobs to finish...
error: linking with `rust-lld` failed: exit status: 1
  |
  = note: "rust-lld" "-flavor" "gnu" "/tmp/rustcjHhWm2/symbols.o" "/home/cquick/atsamx7x-rust/boards/atsame70_xpro/target/thumbv7em-none-eabihf/debug/examples/blinky_irq-59e0ab1a23c855ac.atsamx7x_hal-d5a980c4e9a39481.101fkh0e4q7472nd.rcgu.o.rcgu.o" "--as-needed" "-L" "/home/cquick/atsamx7x-rust/boards/atsame70_xpro/target/thumbv7em-none-eabihf/debug/deps" "-L" "/home/cquick/atsamx7x-rust/boards/atsame70_xpro/target/debug/deps" "-L" "/home/cquick/atsamx7x-rust/boards/atsame70_xpro/target/thumbv7em-none-eabihf/debug/build/atsame70q21b-99c3868bec83c264/out" "-L" "/home/cquick/atsamx7x-rust/boards/atsame70_xpro/target/thumbv7em-none-eabihf/debug/build/cortex-m-bb856671e465605e/out" "-L" "/home/cquick/atsamx7x-rust/boards/atsame70_xpro/target/thumbv7em-none-eabihf/debug/build/cortex-m-rt-6a7ae1b79b44e4b7/out" "-L" "/home/cquick/.rustup/toolchains/1.63.0-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib" "-Bstatic" "/tmp/rustcjHhWm2/libcortex_m_rt-a8a12616f111b1f5.rlib" "--start-group" "/tmp/rustcjHhWm2/libcortex_m-aaa8d6483869b2d6.rlib" "--end-group" "/home/cquick/.rustup/toolchains/1.63.0-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib/libcompiler_builtins-814a2ee6da8588fb.rlib" "-Bdynamic" "--eh-frame-hdr" "-znoexecstack" "-L" "/home/cquick/.rustup/toolchains/1.63.0-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib" "-o" "/home/cquick/atsamx7x-rust/boards/atsame70_xpro/target/thumbv7em-none-eabihf/debug/examples/blinky_irq-59e0ab1a23c855ac" "--gc-sections" "-Tlink.x"
  = note: rust-lld: error: undefined symbol: _SEGGER_RTT
          >>> referenced by lib.rs:175 (/home/cquick/.cargo/registry/src/github.com-1ecc6299db9ec823/rtt-target-0.3.1/src/lib.rs:175)
          >>>               /home/cquick/atsamx7x-rust/boards/atsame70_xpro/target/thumbv7em-none-eabihf/debug/examples/blinky_irq-59e0ab1a23c855ac.atsamx7x_hal-d5a980c4e9a39481.101fkh0e4q7472nd.rcgu.o.rcgu.o:(rtt_target::UpChannel::conjure::hb1d03a52c6959c14)
          >>> referenced by lib.rs:175 (/home/cquick/.cargo/registry/src/github.com-1ecc6299db9ec823/rtt-target-0.3.1/src/lib.rs:175)
          >>>               /home/cquick/atsamx7x-rust/boards/atsame70_xpro/target/thumbv7em-none-eabihf/debug/examples/blinky_irq-59e0ab1a23c855ac.atsamx7x_hal-d5a980c4e9a39481.101fkh0e4q7472nd.rcgu.o.rcgu.o:(rtt_target::UpChannel::conjure::hb1d03a52c6959c14)
          >>> referenced by lib.rs:181 (/home/cquick/.cargo/registry/src/github.com-1ecc6299db9ec823/rtt-target-0.3.1/src/lib.rs:181)
          >>>               /home/cquick/atsamx7x-rust/boards/atsame70_xpro/target/thumbv7em-none-eabihf/debug/examples/blinky_irq-59e0ab1a23c855ac.atsamx7x_hal-d5a980c4e9a39481.101fkh0e4q7472nd.rcgu.o.rcgu.o:(rtt_target::UpChannel::conjure::hb1d03a52c6959c14)
          >>> referenced 5 more times

error: could not compile `atsame70_xpro` due to previous error
```

```
Default host: x86_64-unknown-linux-gnu
rustup home:  /home/cquick/.rustup

installed toolchains
--------------------

stable-x86_64-unknown-linux-gnu (default)
1.63.0-x86_64-unknown-linux-gnu
1.69.0-x86_64-unknown-linux-gnu

installed targets for active toolchain
--------------------------------------

thumbv7em-none-eabihf
x86_64-unknown-linux-gnu

active toolchain
----------------

1.63.0-x86_64-unknown-linux-gnu (overridden by '/home/cquick/atsamx7x-rust/rust-toolchain.toml')
rustc 1.63.0 (4b91a6ea7 2022-08-08)
```

michalfita, 2023-05-06:
I've just run `cargo build` on the tip of `development` branch and I don't see any issues. That's with Rust 1.63 as well.

What branch / version you're trying to build?

michalfita, 2023-05-06:
Try `cargo update` first.

cquick01, 2023-05-07:
Sorry for not including more info. Now seeing the same issue on 2 separate machines. Both Ubuntu 22.04 running under WSL2. Will try on bare metal.

Running off `development`. Just trying to follow the two commands in the README.

```
➜  ~/atsamx7x-rust git:(development) $ rustup target add thumbv7em-none-eabihf
info: component 'rust-std' for target 'thumbv7em-none-eabihf' is up to date

➜  ~/atsamx7x-rust git:(development) $ cd boards/atsame70_xpro

➜  ~/atsamx7x-rust/boards/atsame70_xpro git:(development) $ git status
On branch development
Your branch is up to date with 'origin/development'.

nothing to commit, working tree clean

➜  ~/atsamx7x-rust/boards/atsame70_xpro git:(development) $ git --no-pager log -n1
commit 8b3258d33c1339eae4400817595aec2b540829cd (HEAD -> development, origin/development, origin/HEAD)
Author: Gabriel Górski <glaeqen@gmail.com>
Date:   Mon Jan 2 23:45:05 2023 +0100

    can: impl for `mcan` crate

    Adds an implementation for the mcan crate.

    Refer to #51.

➜  ~/atsamx7x-rust/boards/atsame70_xpro git:(development) $ cargo build --examples
    Updating crates.io index
   Compiling proc-macro2 v1.0.47
   Compiling quote v1.0.21
   Compiling unicode-ident v1.0.5
   Compiling syn v1.0.103
   Compiling semver-parser v0.7.0
   Compiling version_check v0.9.4
   Compiling cortex-m v0.7.6
   Compiling nb v1.0.0
   Compiling void v1.0.2
   Compiling vcell v0.1.3
   Compiling semver v1.0.14
   Compiling autocfg v1.1.0
   Compiling bitfield v0.13.2
   Compiling cortex-m-rt v0.7.1
   Compiling rustversion v1.0.9
   Compiling paste v1.0.9
   Compiling hashbrown v0.12.3
   Compiling heck v0.4.0
   Compiling byteorder v1.4.3
   Compiling atsame70q21b v0.25.0 (/home/cquick/atsamx7x-rust/pac/atsame70q21b)
   Compiling ufmt-write v0.1.0
   Compiling atsamx7x-hal v0.4.2 (/home/cquick/atsamx7x-rust/hal)
   Compiling stable_deref_trait v1.2.0
   Compiling rtic-monotonic v1.0.0
   Compiling usb-device v0.2.9
   Compiling bit-iter v1.1.1
   Compiling cfg-if v1.0.0
   Compiling rtic-core v1.0.0
   Compiling bare-metal v1.0.0
   Compiling panic-halt v0.2.0
   Compiling volatile-register v0.2.1
   Compiling nb v0.1.3
   Compiling semver v0.9.0
   Compiling hash32 v0.2.1
   Compiling proc-macro-error-attr v1.0.4
   Compiling proc-macro-error v1.0.4
   Compiling cortex-m-rtic v1.1.3
   Compiling embedded-hal v0.2.7
   Compiling indexmap v1.9.1
   Compiling rustc_version v0.2.3
   Compiling usbd-serial v0.1.1
[... 98 lines cut ...]
    Checking rtic-core v1.0.0
    Checking cfg-if v1.0.0
    Checking panic-halt v0.2.0
    Checking nb v0.1.3
    Checking volatile-register v0.2.1
    Checking fugit v0.3.6
    Checking bare-metal v0.2.5
    Checking cortex-m-rt v0.7.3
    Checking strum v0.24.1
    Checking hash32 v0.2.1
    Checking embedded-hal v0.2.7
    Checking heapless v0.7.16
    Checking cortex-m v0.7.7
    Checking usbd-serial v0.1.1
    Checking mcan-core v0.2.2
    Checking rtt-target v0.3.1
    Checking atsame70q21b v0.25.0 (/home/cquick/atsamx7x-rust/pac/atsame70q21b)
    Checking panic-rtt-target v0.1.2
    Checking cortex-m-rtic v1.1.4
    Checking atsamx7x-hal v0.4.2 (/home/cquick/atsamx7x-rust/hal)
    Checking atsame70_xpro v0.4.2 (/home/cquick/atsamx7x-rust/boards/atsame70_xpro)
    Finished dev [unoptimized + debuginfo] target(s) in 7.76s
```

Thanks for any help

michalfita, 2023-05-07:
Do you have Rust installed on the Windows host as well?

cquick01, 2023-05-07:
Nope, Rust is not installed on the Windows side. Does it need to be? Admittedly I am not nearly as experienced with Rust than say, Python... Thank you for your patience.

The steps I took (in Ubuntu under WSL) are as follows. Are there any dependencies or any other configurations/setup I may be missing?

1. Install Rust with `curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sh`, add `source "$HOME/.cargo/env"` to shell config and restart shell.
2. Clone `atsamx7x-rust`
3. Install the ARMv7-EM target: `rustup target add thumbv7em-none-eabihf`
4. Build examples: `cd boards/atsame70_xpro && cargo build --examples`

I am also going to try building on a Manjaro laptop that I have laying around. I'll share those results soon.

Thanks again!

cquick01, 2023-05-07:
Unfortunately, getting the same error on Manjaro too... Was hoping WSL was the culprit somehow but that doesn't seem to be the case.

`cargo build` actually does seem to pass, with just one warning (minor unused import), but of course doesn't produce the examples bins. But I can't get `cargo build --examples` working anywhere

```
$ pwd
/home/cquick/atsamx7x-rust/boards/atsame70_xpro
$ uname -a
Linux manjaro 6.1.12-1-MANJARO #1 SMP PREEMPT_DYNAMIC Tue Feb 14 21:59:10 UTC 2023 x86_64 GNU/Linux
$ rustup show
Default host: x86_64-unknown-linux-gnu
rustup home:  /home/cquick/.rustup

installed toolchains
--------------------

stable-x86_64-unknown-linux-gnu (default)
1.63.0-x86_64-unknown-linux-gnu

installed targets for active toolchain
--------------------------------------

thumbv7em-none-eabihf
x86_64-unknown-linux-gnu

active toolchain
----------------

1.63.0-x86_64-unknown-linux-gnu (overridden by '/home/cquick/atsamx7x-rust/rust-toolchain.toml')
rustc 1.63.0 (4b91a6ea7 2022-08-08)

```

Cargo build

```
$ cargo build
   Compiling proc-macro2 v1.0.47
   Compiling unicode-ident v1.0.5
   Compiling quote v1.0.21
   Compiling syn v1.0.103
   Compiling semver-parser v0.7.0
   Compiling version_check v0.9.4
   Compiling cortex-m v0.7.6
   Compiling nb v1.0.0
   Compiling void v1.0.2
   Compiling semver v1.0.14
   Compiling vcell v0.1.3
   Compiling autocfg v1.1.0
   Compiling bitfield v0.13.2
   Compiling rustversion v1.0.9
   Compiling cortex-m-rt v0.7.1
   Compiling hashbrown v0.12.3
   Compiling paste v1.0.9
   Compiling heck v0.4.0
   Compiling byteorder v1.4.3
   Compiling atsame70q21b v0.25.0 (/home/cquick/atsamx7x-rust/pac/atsame70q21b)
   Compiling stable_deref_trait v1.2.0
   Compiling rtic-monotonic v1.0.0
   Compiling atsamx7x-hal v0.4.2 (/home/cquick/atsamx7x-rust/hal)
   Compiling usb-device v0.2.9
   Compiling ufmt-write v0.1.0
   Compiling bit-iter v1.1.1
   Compiling cfg-if v1.0.0
   Compiling rtic-core v1.0.0
   Compiling bare-metal v1.0.0
   Compiling panic-halt v0.2.0
   Compiling nb v0.1.3
   Compiling semver v0.9.0
   Compiling volatile-register v0.2.1
[... 44 lines cut ...]
          >>>               /home/cquick/atsamx7x-rust/boards/atsame70_xpro/target/thumbv7em-none-eabihf/debug/examples/blinky_irq-59e0ab1a23c855ac.atsamx7x_hal-d5a980c4e9a39481.101fkh0e4q7472nd.rcgu.o.rcgu.o:(rtt_target::UpChannel::conjure::hb1d03a52c6959c14)
          >>> referenced by lib.rs:175 (/home/cquick/.cargo/registry/src/github.com-1ecc6299db9ec823/rtt-target-0.3.1/src/lib.rs:175)
          >>>               /home/cquick/atsamx7x-rust/boards/atsame70_xpro/target/thumbv7em-none-eabihf/debug/examples/blinky_irq-59e0ab1a23c855ac.atsamx7x_hal-d5a980c4e9a39481.101fkh0e4q7472nd.rcgu.o.rcgu.o:(rtt_target::UpChannel::conjure::hb1d03a52c6959c14)
          >>> referenced by lib.rs:181 (/home/cquick/.cargo/registry/src/github.com-1ecc6299db9ec823/rtt-target-0.3.1/src/lib.rs:181)
          >>>               /home/cquick/atsamx7x-rust/boards/atsame70_xpro/target/thumbv7em-none-eabihf/debug/examples/blinky_irq-59e0ab1a23c855ac.atsamx7x_hal-d5a980c4e9a39481.101fkh0e4q7472nd.rcgu.o.rcgu.o:(rtt_target::UpChannel::conjure::hb1d03a52c6959c14)
          >>> referenced 5 more times
          

error: could not compile `atsame70_xpro` due to previous error
warning: build failed, waiting for other jobs to finish...
error: linking with `rust-lld` failed: exit status: 1
  |
  = note: "rust-lld" "-flavor" "gnu" "/tmp/rustcJHvQtT/symbols.o" "/home/cquick/atsamx7x-rust/boards/atsame70_xpro/target/thumbv7em-none-eabihf/debug/examples/blinky_delay-873d777b61c44224.atsamx7x_hal-d5a980c4e9a39481.101fkh0e4q7472nd.rcgu.o.rcgu.o" "--as-needed" "-L" "/home/cquick/atsamx7x-rust/boards/atsame70_xpro/target/thumbv7em-none-eabihf/debug/deps" "-L" "/home/cquick/atsamx7x-rust/boards/atsame70_xpro/target/debug/deps" "-L" "/home/cquick/atsamx7x-rust/boards/atsame70_xpro/target/thumbv7em-none-eabihf/debug/build/atsame70q21b-99c3868bec83c264/out" "-L" "/home/cquick/atsamx7x-rust/boards/atsame70_xpro/target/thumbv7em-none-eabihf/debug/build/cortex-m-bb856671e465605e/out" "-L" "/home/cquick/atsamx7x-rust/boards/atsame70_xpro/target/thumbv7em-none-eabihf/debug/build/cortex-m-rt-6a7ae1b79b44e4b7/out" "-L" "/home/cquick/.rustup/toolchains/1.63.0-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib" "-Bstatic" "/tmp/rustcJHvQtT/libcortex_m_rt-a8a12616f111b1f5.rlib" "--start-group" "/tmp/rustcJHvQtT/libcortex_m-aaa8d6483869b2d6.rlib" "--end-group" "/home/cquick/.rustup/toolchains/1.63.0-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib/libcompiler_builtins-814a2ee6da8588fb.rlib" "-Bdynamic" "--eh-frame-hdr" "-znoexecstack" "-L" "/home/cquick/.rustup/toolchains/1.63.0-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib" "-o" "/home/cquick/atsamx7x-rust/boards/atsame70_xpro/target/thumbv7em-none-eabihf/debug/examples/blinky_delay-873d777b61c44224" "--gc-sections" "-Tlink.x"
  = note: rust-lld: error: undefined symbol: _SEGGER_RTT
          >>> referenced by lib.rs:175 (/home/cquick/.cargo/registry/src/github.com-1ecc6299db9ec823/rtt-target-0.3.1/src/lib.rs:175)
          >>>               /home/cquick/atsamx7x-rust/boards/atsame70_xpro/target/thumbv7em-none-eabihf/debug/examples/blinky_delay-873d777b61c44224.atsamx7x_hal-d5a980c4e9a39481.101fkh0e4q7472nd.rcgu.o.rcgu.o:(rtt_target::UpChannel::conjure::hb1d03a52c6959c14)
[... cut ...]
          >>> referenced by lib.rs:181 (/home/cquick/.cargo/registry/src/github.com-1ecc6299db9ec823/rtt-target-0.3.1/src/lib.rs:181)
          >>>               /home/cquick/atsamx7x-rust/boards/atsame70_xpro/target/thumbv7em-none-eabihf/debug/examples/blinky_delay-873d777b61c44224.atsamx7x_hal-d5a980c4e9a39481.101fkh0e4q7472nd.rcgu.o.rcgu.o:(rtt_target::UpChannel::conjure::hb1d03a52c6959c14)
          >>> referenced 5 more times
          

error: could not compile `atsame70_xpro` due to previous error
```

michalfita, 2023-05-09:
OK, I confirm the problem. However, the issue lies somewhere in dependencies, not our demo code. @tmplt do you know solution to this problem? BTW. CI doesn't seem to run full build only check, that's why this problem slipped there.

cquick01, 2023-05-10:
Diffing against the atsamv71 ones, looks like they are missing `rtt_target::rtt_init_print!();`. Adding these lines got the builds working

```diff
diff --git a/boards/atsame70_xpro/examples/blinky_delay.rs b/boards/atsame70_xpro/examples/blinky_delay.rs
index be22ad1..f35585a 100644
--- a/boards/atsame70_xpro/examples/blinky_delay.rs
+++ b/boards/atsame70_xpro/examples/blinky_delay.rs
@@ -26,6 +26,7 @@ mod app {
 
     #[init]
     fn init(ctx: init::Context) -> (Shared, Local, init::Monotonics) {
+        rtt_target::rtt_init_print!();
         let clocks = Tokens::new(
             (ctx.device.PMC, ctx.device.SUPC, ctx.device.UTMI),
             &ctx.device.WDT.into(),
diff --git a/boards/atsame70_xpro/examples/blinky_irq.rs b/boards/atsame70_xpro/examples/blinky_irq.rs
index 55db85e..399023f 100644
--- a/boards/atsame70_xpro/examples/blinky_irq.rs
+++ b/boards/atsame70_xpro/examples/blinky_irq.rs
@@ -26,6 +26,7 @@ mod app {
 
     #[init]
     fn init(ctx: init::Context) -> (Shared, Local, init::Monotonics) {
+        rtt_target::rtt_init_print!();
         let clocks = Tokens::new(
             (ctx.device.PMC, ctx.device.SUPC, ctx.device.UTMI),
             &ctx.device.WDT.into(),
```

Now unfortunately I'm having trouble getting my same70 board to run these binaries so I can't actually test it, but that's a separate issue I'll have to work out.

michalfita, 2023-07-30:
@cquick01 any progress since? Maybe you could PR the changes?

cquick01, 2023-07-31:
I was never able to get the binaries to run. Haven't really looked at it in a month or two.

PR #70 (2023-12-22): Fix Broken Examples
This resolves #54 .
closed 2023-12-22 by michalfita