atsams-rs/atsamx7x-rust #63: rustc 1.72.0: flooded with `warning: lint `const_err` has been removed: converted into hard error`

michalfita, 2023-09-14:
```
   Compiling atsamv71q19b v0.25.0 (/home/manveru/Projects/atsamx7x-rust/pac/atsamv71q19b)
warning: lint `const_err` has been removed: converted into hard error, see issue #71800 <https://github.com/rust-lang/rust/issues/71800> for more information
 --> pac/atsams70n19b/src/lib.rs:3:9
  |
3 | #![deny(const_err)]
  |         ^^^^^^^^^
  |
  = note: `#[warn(renamed_and_removed_lints)]` on by default
```
So, it's materialization of https://github.com/rust-lang/rust/issues/71800.

We need CI to actually test with MRV and latest.

tmplt, 2023-09-15:
> We need CI to actually test with MRV and latest.
This is a good idea. Minimum would be a required to pass, but any errors from latest could be emitted as warnings for review in PRs.

michalfita, 2023-09-17:
> any errors from latest could be emitted as warnings for review

Except GitHub contrary to GitLab cannot emit warning statues from builds. In general I find GitLab superior to GitHub, it's just much less popular choice for Rust's crates.

tmplt, 2023-09-17:
I must have conflated the two.

martinmortsell, 2023-10-10:
With the merge of #64 this issue is resolved, right?

michalfita, 2023-10-11:
Yes, it is. We now need to move to PACs generated with newer `svd2rust` what I already exercised.

closed 2023-10-11 by michalfita