atsams-rs/atsamx7x-rust #65: Example code for ADC on Xult (blue) PCB is making use of AFEC0_AD6 which is not exposed

blietaer, 2023-11-28:
Mmmh quite an interesting choice to run [ADC test/example](https://github.com/atsams-rs/atsamx7x-rust/blob/development/boards/atsamv71_xult/examples/periodic_adc_sample.rs) code on the AFE0_AD6 channel...as this pin is not (easily) exposed on the [SAM V71 Xultra (blue) PCB dev. kit](https://www.microchip.com/en-us/development-tool/atsamv71-xult): what am I missing here ? 🤔

Well, I just tested it on the Xult/EXT-1 PIN#4 (PA19/AFEC0_AD8), working fine: shall I commit the change ?

michalfita, 2023-11-28:
@tmplt These examples are yours. Could you comment on that, please?

tmplt, 2023-11-28:
I have little recollection of this choice. It may be differences in hardware revisions. If it is outdated, do PR a change.

commit ec0a3982 (2023-11-28): [Closes #65] Merge pull request #66 from blietaer/xult-adc-example-code

ADC Example tested on XULT-V71 with EXT-1 PIN#4 (PA19/AFEC0_AD8)
closed 2023-11-28 by michalfita