atsams-rs/atsamx7x-rust #75: Panicked at examples/uart_loopback_test.rs

blietaer, 2024-01-29:
Dear SAM Team,

I am afraid the UART examples are not working out of the box on the [SAM V71 Xultra (blue) Board](https://www.microchip.com/en-us/development-tool/atsamv71-xult) anymore (? I can't confirm it ever did, first time I try it).
Maybe I am doing something wrong, or oversight of a minimum (undocumented) wiring here ?

[image]
Sounds like a Clock issue as we can easily track this error msg down to `uart.rs:358` (might want to try and play with both `MCK` and `PCK` values).

I have the (exact) same situation for both the `examples/uart_loopback_test.rs` and the `examples/uart_interrupts`, while both `examples/usart_loopback.rs` (using a small wiring PB00-PB01 on J505) and `examples/usb_serial_echo.rs` (using second 9target) USB port) are working just fine.

Any hint/correction/suggestion is welcome here ! :)

Cheers,
Ben

michalfita, 2024-01-29:
@blietaer for clarity as you haven't mentioned. Is it on released version (which one?) or from top of `development` branch?

Maybe @martinmortsell would have a quick hint for you?

blietaer, 2024-01-29:
Wow many thanks to both of you for this super quick feedback.

> @blietaer for clarity as you haven't mentioned: Is it on released version (which one?) or from top of `development` branch?
Ah yes, this is `development` branch, sorry.
 
> Have you changed anything in the example code anywhere else?
No no, this is a vanilla clone from repo.
 
> The problem is that there's a requirement that mck needs to be at least 3 times bigger than pck, and this is not fulfilled as the prescaler value is set to 1. Changing the HccPrescaler should fix this issue.
Yes, this is my understanding just reviewing `uart.rs`: mmmh let me check if I can play with `HccPrescaler` directly from the example code, without touching anything under the HAL/PAC hood.

blietaer, 2024-01-30:
Changing the `pres` value (second parameter of `configure()` ) from `1` to `3` seems indeed to fix this: 
[image]

michalfita, 2024-01-30:
@blietaer PR this change, please.

blietaer, 2024-01-30:
Will do, but actually I was a bit quick: this only solves the `example\uart_interrupts.rs` but not the `example\uart_loopback_test`: I can't have anything printed (and I wired the same) not sure if my modification is suspicious here...can't hardly debug it :/
Trying different Tx/Rx pin (loop-wired too..), but not joy...

closed 2025-10-27 by michalfita